stereolabs/zed-ros-wrapper #942: ZED-ROS-ERROR:Invalid calibration file

AutomationForever, 2024-04-15:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

ERROR: Invalid calibration file. 
On Linux, call export LC_ALL="C" and restart your application. 
ZED (Init) >> Wrong value in calibration file. Downloading it

### Use-Case

I can't start my camera by "roslaunch zed-wrapper zed.launch"

### Anything else?

Please help me to solve it.

Myzhar, 2024-04-15:
Hi @AutomationForever 
is the PC connected to the internet?
Are you using Docker?

AutomationForever, 2024-04-19:
__

> Hi @AutomationForever is the PC connected to the internet? Are you using Docker?

Thank you for reminding me. When my device is connected to the Internet,  it starts working properly.

Myzhar, 2024-04-19:
If you are using Docker you should make a shared folder to store the setting files on the host. In this way you only download them once.
Read more here:
https://github.com/stereolabs/zed-ros2-wrapper/tree/master/docker#volumes

closed 2024-10-03 by Myzhar