stereolabs/zed-ros-wrapper #948: roslaunch hangs on "* Waiting for valid static transformations..."

amstrudy, 2024-05-23:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I just installed the ZED SDK and ROS wrapper on my Ubuntu 20.04 machine in order to collect some bags from my ZED2i. I installed the ZED SDK 4.1.1, and the ROS wrapper into a catkin workspace. Building was no issue--it was only upon running `roslaunch zed_wrapper zed2i.launch` that I had an issue. Everything seems to run normally but then it hangs forever on "Waiting for valid static transformations..."

### Steps to Reproduce

1. Install ZED SDK 4.1.1
2. Clone `zed_ros_wrapper`
3. `catkin build -DCMAKE_BUILD_TYPE=Release`
4. `roslaunch zed_wrapper zed2i.launch`

### Expected Result

I'm expecting the topics to start publishing, but they don't because the startup is hanging at this valid static transforms step.

### Actual Result

Output of roslaunch:
```
started roslaunch server http://anjashep-frog-lab:35391/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /zed2i/zed2i_description: <?xml version="1....
 * /zed2i/zed_node/auto_exposure_gain: True
 * /zed2i/zed_node/auto_whitebalance: True
 * /zed2i/zed_node/brightness: 4
 * /zed2i/zed_node/contrast: 4
 * /zed2i/zed_node/depth/depth_mode: ULTRA
 * /zed2i/zed_node/depth/depth_stabilization: 1
 * /zed2i/zed_node/depth/max_depth: 15.0
 * /zed2i/zed_node/depth/min_depth: 0.3
 * /zed2i/zed_node/depth/openni_depth_mode: False
 * /zed2i/zed_node/depth_confidence: 50
 * /zed2i/zed_node/depth_texture_conf: 100
 * /zed2i/zed_node/exposure: 100
 * /zed2i/zed_node/gain: 100
 * /zed2i/zed_node/gamma: 8
 * /zed2i/zed_node/general/base_frame: base_link
 * /zed2i/zed_node/general/camera_flip: AUTO
 * /zed2i/zed_node/general/camera_model: zed2i
 * /zed2i/zed_node/general/camera_name: zed2i
 * /zed2i/zed_node/general/gpu_id: -1
 * /zed2i/zed_node/general/grab_frame_rate: 30
 * /zed2i/zed_node/general/grab_resolution: HD720
 * /zed2i/zed_node/general/pub_downscale_factor: 2.0
 * /zed2i/zed_node/general/pub_frame_rate: 15.0
 * /zed2i/zed_node/general/pub_resolution: CUSTOM
 * /zed2i/zed_node/general/region_of_interest: []
 * /zed2i/zed_node/general/sdk_verbose: 1
 * /zed2i/zed_node/general/self_calib: True
 * /zed2i/zed_node/general/serial_number: 0
 * /zed2i/zed_node/general/svo_compression: 2
 * /zed2i/zed_node/general/svo_realtime: True
 * /zed2i/zed_node/general/zed_id: 0
 * /zed2i/zed_node/hue: 0
 * /zed2i/zed_node/mapping/clicked_point_topic: /clicked_point
 * /zed2i/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed2i/zed_node/mapping/mapping_enabled: False
 * /zed2i/zed_node/mapping/max_mapping_range: -1
 * /zed2i/zed_node/mapping/resolution: 0.05
[... 214 lines cut ...]
[ INFO] [1716472243.218868478]:  * Advertised on service /zed2i/zed_node/start_svo_recording
[ INFO] [1716472243.219827500]:  * Advertised on service /zed2i/zed_node/stop_svo_recording
[ INFO] [1716472243.220894816]:  * Advertised on service /zed2i/zed_node/set_led_status
[ INFO] [1716472243.221873117]:  * Advertised on service /zed2i/zed_node/toggle_led
[ INFO] [1716472243.222779914]:  * Advertised on service /zed2i/zed_node/start_remote_stream
[ INFO] [1716472243.223605528]:  * Advertised on service /zed2i/zed_node/stop_remote_stream
[ INFO] [1716472243.224499665]:  * Advertised on service /zed2i/zed_node/set_roi
[ INFO] [1716472243.225385450]:  * Advertised on service /zed2i/zed_node/set_roi
[ INFO] [1716472243.225637906]: +++ ZED Node started +++
[ INFO] [1716472243.225883516]: Published static transform 'zed2i_imu_link' -> 'zed2i_left_camera_frame'
[ INFO] [1716472243.226120783]: *** Starting Positional Tracking ***
[ INFO] [1716472243.226206772]:  * Waiting for valid static transformations...
```

### ZED Camera model

ZED

### Environment

```shell
OS: Ubuntu 20.04
ZED SDK: 4.1.1
ROS Noetic
```

Myzhar, 2024-05-23:
Hi @amstrudy 
can you please send me the TF tree? (you can use rtq to save it as an image)

amstrudy, 2024-05-29:
@Myzhar Somehow this resolved itself after a few days... not sure what the issue was but I can't reproduce the problem now.

closed 2024-05-29 by amstrudy