stereolabs/zed-ros-wrapper #949: separate downscale for depth and color (was available before)

bmegli, 2024-06-10:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

When moving to new driver "backend" following ROS 2 implementation, possibility to use different down-scaling for depth and color data was removed.

https://github.com/stereolabs/zed-ros-wrapper/blob/37d0c0224174e64b9676e4900e33d1a2852685e3/zed_wrapper/params/common.yaml#L31-L32

The proposal is to have this possibility back

### Use-Case

In non-local scenarios:
- depth data compression, processing, bandwidth requirements are high
- RGB compression/processing/bandwidth requirements are low

In the past it was possible to use high resolution RGB video data with lower resolution depth data (downsampled independently from RGB/video)

Myzhar, 2024-06-10:
This feature has been removed for data synchronization issues.
Adding it back is not planned.

The repository is open-source, so I recommend forking it to bring forward your custom version if you need this feature.

bmegli, 2024-06-10:
Umm ok!

closed 2024-06-10 by bmegli