stereolabs/zed-ros-wrapper #953: Positional Tracking in ROS wrapper ZED2i

LyLewis, 2024-08-22:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

As i want to use ZED2i for mapping around my robot, the positional tracking is vital for everything. When i initialize and start the ROS node, everything for the tracking(pose, path map...) is normal, tracking state is acceptable as well, but the pose jumped far away after the camera stay statics for a while.,that's wired. i doubt that it is cased by the IMU fusion noise. would you have some idea for that?

### Use-Case

i want a accurate positional tracking in ZED2i;
platform i used:
Jetson Orin NX 16GB;
ZED2i;
Jetpack 5.1.3;
CUDA 11.4;
ROS noetic

closed 2025-02-12 by LyLewis

illuminate97, 2025-05-03:
Hello LyLewis,

i am facing exactly the same problem. Were you able to fix it?

## Environment
Jetson AGX Orin 32GB;
ZED2i;
Jetpack 5.1.2;
CUDA 11.8;
ROS noetic

Myzhar, 2025-05-03:
We recommend to migrate to ROS 2 and use Positional Tracking GEN_2 with the ZED SDK v5.0

illuminate97, 2025-05-03:
Hello Myzhar,

i already upgraded to SDK Version 5.0 and enabled GEN_2 tracking. Unfortunately, upgrading to ROS2 is currently not an option, because everything else relies on ROS Noetic. :/ 

Here is the configuration:

process[zed2i/zed_node-3]: started with pid [332170]
[INFO] [1746271676.410432131]: Initializing nodelet with 12 worker threads.
[INFO] [1746271676.620166252]: ********** Starting nodelet '/zed2i/zed_node' **********
**[INFO] [1746271676.620353582]: SDK version: 5.0.0**
[INFO] [1746271676.620466672]: *** GENERAL PARAMETERS ***
[INFO] [1746271676.621349692]:  * Camera Model by param	-> zed2i
[INFO] [1746271676.622045702]:  * Camera Name			-> zed2i
[INFO] [1746271676.622731216]:  * Camera resolution		-> HD720
[INFO] [1746271676.623365625]:  * Camera Grab Framerate	-> 30
[INFO] [1746271676.624222373]:  * Grab compute capping FPS	-> 15 Hz
[INFO] [1746271676.625043632]:  * Publishing resolution	-> CUSTOM
[INFO] [1746271676.625815675]:  * Publishing downscale factor	-> 2
[INFO] [1746271676.626578566]:  * Gpu ID			-> -1
[INFO] [1746271676.627330289]:  * Camera ID			-> 0
[INFO] [1746271676.628077691]:  * SDK Verbose level		-> 1
[INFO] [1746271676.628883846]:  * Camera Flip			-> AUTO
[INFO] [1746271676.629817812]:  * Self calibration		-> ENABLED
[INFO] [1746271676.631388490]:  * Region of interest		-> []
Ci sei o no?
[INFO] [1746271676.631515340]: *** DYNAMIC PARAMETERS (Init. values) ***
[INFO] [1746271676.632366359]:  * [DYN] brightness		-> 4
[INFO] [1746271676.633208323]:  * [DYN] contrast		-> 4
[INFO] [1746271676.634018383]:  * [DYN] hue			-> 0
[INFO] [1746271676.634812826]:  * [DYN] saturation		-> 4
[INFO] [1746271676.635613029]:  * [DYN] sharpness		-> 4
[INFO] [1746271676.636436625]:  * [DYN] gamma			-> 8
[INFO] [1746271676.637246748]:  * [DYN] auto_exposure_gain	-> ENABLED
[INFO] [1746271676.639379226]:  * [DYN] auto_whitebalance	-> ENABLED
[INFO] [1746271676.640889487]:  * [DYN] Depth confidence	-> 50
[INFO] [1746271676.641682619]:  * [DYN] Depth texture conf.	-> 100
[INFO] [1746271676.642479430]:  * [DYN] point_cloud_freq	-> 10 Hz
[INFO] [1746271676.642550247]: *** DEPTH PARAMETERS ***
[INFO] [1746271676.643459828]:  * Depth mode			-> ULTRA [3]
[INFO] [1746271676.644306112]:  * OpenNI mode			-> ENABLED
[INFO] [1746271676.645118187]:  * Depth Stabilization		-> 1
[INFO] [1746271676.645806453]:  * Minimum depth		-> 0.3 m
[INFO] [1746271676.646215706]:  * Maximum depth		-> 15 m
[INFO] [1746271676.646259899]: *** POSITIONAL TRACKING PARAMETERS ***
[INFO] [1746271676.646650657]:  * Positional tracking		-> ENABLED
**[INFO] [1746271676.647202536]:  * Positional tracking mode	-> GEN 2**
[INFO] [1746271676.647721904]:  * Set gravity as origin	-> ENABLED
[INFO] [1746271676.648609308]:  * Path rate			-> 2 Hz
[INFO] [1746271676.649449704]:  * Path history size		-> INFINITE
[INFO] [1746271676.651071903]:  * Odometry DB path		-> 
[INFO] [1746271676.651903371]:  * Save Area Memory on closing	-> DISABLED
[INFO] [1746271676.652747606]:  * Area Memory			-> ENABLED
[INFO] [1746271676.653494881]:  * IMU Fusion			-> ENABLED
[INFO] [1746271676.654218315]:  * Floor alignment		-> DISABLED
[INFO] [1746271676.654951317]:  * Init Odometry with first valid pose data -> ENABLED
[INFO] [1746271676.655665696]:  * Force 2D mode		-> DISABLED
[INFO] [1746271676.656404682]:  * Broadcast odometry TF	-> ENABLED
[INFO] [1746271676.657147508]:  * Broadcast map pose TF	-> ENABLED
[INFO] [1746271676.657867199]:  * Camera is static		-> DISABLED
[INFO] [1746271676.658594473]:  * Depth minimum range		-> 0
[INFO] [1746271676.658664906]: *** MAPPING PARAMETERS ***
[INFO] [1746271676.659348147]:  * Mapping			-> DISABLED
[INFO] [1746271676.660017565]:  * Clicked point topic		-> /clicked_point
[INFO] [1746271676.660072766]: *** OBJECT DETECTION PARAMETERS ***
[INFO] [1746271676.660759591]:  * Object Detection		-> DISABLED
[INFO] [1746271676.661446257]:  * Object confidence		-> 50
[INFO] [1746271676.662124698]:  * Object tracking		-> ENABLED
[INFO] [1746271676.662792452]:  * Detection max range		-> 15
[INFO] [1746271676.663470573]:  * Allow reduced precision	-> ENABLED
[INFO] [1746271676.664156407]:  * Prediction Timeout		-> 0.5
[... 8 lines cut ...]
[INFO] [1746271676.670237581]: *** SENSORS PARAMETERS ***
[INFO] [1746271676.671048504]:  * Sensors timestamp sync	-> DISABLED
[INFO] [1746271676.672140391]:  * Max sensors rate		-> 200
[INFO] [1746271676.672850897]:  * Broadcast IMU pose TF	-> ENABLED
[INFO] [1746271676.672917586]: *** SVO PARAMETERS ***
[INFO] [1746271676.673555803]:  * SVO input file: 		-> 
[INFO] [1746271676.674239077]:  * SVO REC compression		-> H265
[INFO] [1746271676.674884558]: *** COORDINATE FRAMES ***
[INFO] [1746271676.676680711]:  * camera_frame			-> zed2i_camera_center
[INFO] [1746271676.676740552]:  * imu_link			-> zed2i_imu_link
[INFO] [1746271676.676782729]:  * left_camera_frame		-> zed2i_left_camera_frame
[INFO] [1746271676.676814889]:  * left_camera_optical_frame	-> zed2i_left_camera_optical_frame
[INFO] [1746271676.676854090]:  * right_camera_frame		-> zed2i_right_camera_frame
[INFO] [1746271676.676886730]:  * right_camera_optical_frame	-> zed2i_right_camera_optical_frame
[INFO] [1746271676.676923851]:  * map_frame			-> map
[INFO] [1746271676.676959019]:  * odometry_frame		-> odom
[INFO] [1746271676.676994220]:  * base_frame			-> base_link
[INFO] [1746271676.677032044]:  * depth_frame			-> zed2i_left_camera_frame
[... cut ...]
[INFO] [1746271676.677103629]:  * disparity_frame		-> zed2i_left_camera_frame
[INFO] [1746271676.677139886]:  * disparity_optical_frame	-> zed2i_left_camera_optical_frame
[INFO] [1746271676.677173678]:  * confidence_frame		-> zed2i_left_camera_frame
[INFO] [1746271676.677206639]:  * confidence_optical_frame	-> zed2i_left_camera_optical_frame
[INFO] [1746271676.677271216]: Openni depth mode activated -> Units: mm, Encoding: TYPE_16UC1
[INFO] [1746271676.685430467]:  * Camera coordinate system	-> RIGHT HANDED Z UP X FWD

LyLewis, 2025-06-10:
Hi, illuminate97,
Just set the  Positional tracking mode as "GEN1" and everything is right for me.