stereolabs/zed-ros-wrapper #955: Compilation fails with catkin_make -DCMAKE_BUILD_TYPE=Release

Anchor1566, 2024-09-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I'm experiencing an issue when I try to build my package using `catkin_make -DCMAKE_BUILD_TYPE=Release`. The error seems to be related to the `zed_wrapper_node` component within the zed-ros-wrapper package.

The specific error messages I receive are as follows:
```
-- The C compiler identification is GNU 12.3.0
-- The CXX compiler identification is GNU 12.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/zbx/Desktop/code/SDU/zed_ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/zbx/Desktop/code/SDU/zed_ros_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 4 packages in topological order:
-- ~~  - zed_interfaces
-- ~~  - zed_nodelets
-- ~~  - zed_ros (metapackage)
-- ~~  - zed_wrapper
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'zed_interfaces'
-- ==> add_subdirectory(zed-ros-wrapper/zed-ros-interfaces)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- zed_interfaces: 13 messages, 17 services
-- +++ processing catkin package: 'zed_nodelets'
-- ==> add_subdirectory(zed-ros-wrapper/zed_nodelets)
-- Looking for sgemm_
-- Looking for sgemm_ - not found
-- Looking for sgemm_
-- Looking for sgemm_ - found
-- Found BLAS: /usr/lib/x86_64-linux-gnu/libopenblas.so  
-- Found CUDA: /usr/local/cuda-12.4 (found suitable version "12.4", minimum required is "12") 
-- Found CUDA: /usr/local/cuda-12.4 (found version "12.4") 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found PythonInterp: /usr/bin/python3 (found version "3.10.12") 
-- +++ processing catkin metapackage: 'zed_ros'
-- ==> add_subdirectory(zed-ros-wrapper/zed_ros)
CMake Deprecation Warning at zed-ros-wrapper/zed_ros/CMakeLists.txt:1 (cmake_minimum_required):
[... 346 lines cut ...]
[ 90%] Generating C++ code from zed_interfaces/start_object_detection.srv
[ 91%] Generating C++ code from zed_interfaces/save_area_memory.srv
[ 92%] Generating Javascript code from zed_interfaces/reset_roi.srv
[ 93%] Generating C++ code from zed_interfaces/set_roi.srv
[ 94%] Generating C++ code from zed_interfaces/reset_roi.srv
[ 94%] Built target zed_interfaces_generate_messages_nodejs
[ 95%] Generating Python srv __init__.py for zed_interfaces
[ 96%] Generating Python msg __init__.py for zed_interfaces
[ 96%] Built target zed_interfaces_generate_messages_cpp
[ 96%] Built target zed_interfaces_generate_messages_eus
[ 96%] Built target zed_interfaces_generate_messages_py
make: *** [Makefile:146: all] Error 2
Invoking "make -j12 -l12" failed

### ZED Camera model

ZED

### Environment

```shell
OS： ubuntu 22.04
ros: ros Noetic
ZED SDK: 4.1.3
```

Myzhar, 2024-09-09:
Hi @Anchor1566 
how did you install ROS Noetic on Ubuntu 22.04?
The latest officially supported Ubuntu version is 20.04, so we cannot assure that the ZED ROS Wrapper packages work correctly on Ubuntu 22.04.

Anchor1566, 2024-09-09:
> Hi @Anchor1566 how did you install ROS Noetic on Ubuntu 22.04? The latest officially supported Ubuntu version is 20.04, so we cannot assure that the ZED ROS Wrapper packages work correctly on Ubuntu 22.04.

I install ros noetic in Ubuntu 22.04 in this way，https://github.com/tinkerfuroc/ros_noetic_on_jammy.so, is it best to use ROS in Ubuntu 20.04 or ROS 2 in Ubuntu 22.04 to avoid unknown errors?

Myzhar, 2024-09-09:
The main problem is the library compatibility.
The error that you are facing while compiling the ZED ROS Wrapper is caused by the Boost library and I suppose it's a compatibility issue.
If you use [Ubuntu 20.04 and you install ROS from `apt`](https://wiki.ros.org/noetic/Installation/Ubuntu) you can be sure that you will not face compatibility issues.

closed 2024-10-15 by github-actions[bot]